ros-industrial-consortium/scan_n_plan_workshop #11: Issue- Connection between Nodes

hakan6710, 2022-07-20:
Hello,

i started the nodes as explained in the readme. I cant use the simulation. After pressing ,,Get Detailed Scan", it fails with the following message:

process[joint_trajectory_action-5]: started with pid [1388697]
[ WARN] [1658326459.116264607]: Trajectory state not received for 1.000000 seconds
[ERROR] [1658326589.212103200]: Failed to connect to server, rc: -1. Error: 'Connection timed out' (errno: 110)
[ERROR] [1658326589.212125117]: Failed to connect to server, rc: -1. Error: 'Connection timed out' (errno: 110)
[ERROR] [1658326589.212192764]: Failed to connect to server, rc: -1. Error: 'Connection timed out' (errno: 110)
[ WARN] [1658326589.213150786]: Unable to read velocity limits from 'robot_description' param.  Velocity validation disabled.
[FATAL] [1658326589.216690836]: Failed to connect

I have the default version of "motoman.git". The given instruction did not work for me, because the branch could not be found. Could that be the problem? Which version of it is needed?

marip8, 2022-07-22:
> Is there a more detailed documentation of the workshop or any videos?

[Here is a link to the recorded community meeting](https://www.youtube.com/watch?v=GgTxvIaekjE) where I discussed how this repository works and showed a demo

hakan6710, 2022-07-27:
Hello gavanderhoorn and marip8,

thank you for your replies. I had a misunderstanding of the motoman_driver , i thought it was also able to simulate.

I tried the command  suggested for the real way of simulating and had a problem. ,,Generate Motion Plan" did not work. 

My usage:
1.launch node
2.,,get detailed scan"
3. ,,generate tool path plan"
4.,,get motion plan" -> fails

Log: 
,,
snp_motion_planning_node-7] Error: Discrete collision detected between 'link_4_r' and 'table' with distance -0.0330511
[snp_motion_planning_node-7]
[snp_motion_planning_node-7] at line 148 in /home/roslab/Desktop/sndp_workspace/src/tesseract/tesseract_environment/src/utils.cpp
[snp_motion_planning_node-7] Error: Discrete collision detected between 'ee' and 'link_4_r' with distance -0.0663848
[snp_motion_planning_node-7]
[snp_motion_planning_node-7] at line 148 in /home/roslab/Desktop/sndp_workspace/src/tesseract/tesseract_environment/src/utils.cpp
[snp_motion_planning_node-7] Error: Discrete collision detected between 'link_4_r' and 'scan' with distance -0.0537901"

" Waiting to calibrate...
Observation service is not available
Sending scan approach motion goal
execute scan approach completed!
Waiting to start reconstruction...
Successfully executed scan approach motion
start reconstruction completed!
Waiting to execute scan...
Sending scan trajectory goal
execute scan completed!
Waiting to stop reconstruction...
Successfully executed scan motion
stop reconstruction completed!
Waiting to execute scan departure...
Sending scan departure motion goal
Successfully completed scan and surface reconstruction
execute scan departure completed!
Waiting to plan tool paths...
plan tool paths completed!
Waiting to perform motion planning...
Failed to create motion plan
perform motion planning failed
Waiting to attempt again..."

hakan6710, 2022-07-27:
Am I missing somethin else? 

Thank you for your help in advance.

marip8, 2022-07-27:
We made a number of changes in #5 that dramatically improve planning success and control of the tool path. I would suggest building off that branch and trying again. I think most of the motion planning failures due to collision are caused by the robot reaching waypoints that are too close to the table. We get around this by selecting a sub-region on top of the mesh for tool path planning, rather than generating a tool path over the full extent of the mesh. 

You could also replace the simulation mesh with a mesh of your own (using [this argument](https://github.com/ros-industrial-consortium/scan_n_plan_workshop/blob/8309bca66bce836766760de6ee97b5e0457f4b82/snp_bringup_ros2/launch/start.launch.xml#L5)) where a tool path generated on that mesh won't put the robot in collision.

marip8, 2023-06-05:
Closing; feel free to re-open if this issue is still unresolved

closed 2023-06-05 by marip8