ros-industrial-consortium/scan_n_plan_workshop #18: error when running the simulation

zhiyong-009, 2023-03-12:
after i scan, generate the tool paths and attempt to execute motion planning there are these 2 error types: 

RRTConnect: Motion planning start tree could not be initialized!

at line 207 in /tmp/binarydeb/ros-foxy-onpl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.

RRT: There are no valid initial states!
at Utne 112 in /tnp/binarydeb/ros-foxy-onpl-1.5.0/src/ompl/geonetric/planners/rrt/src/RRT.cpp
RRT: There are no valid initial states!
at line 112 in /tnp/binarydeb/ros-foxy-onpl-1.5.0/src/ompl/geonetric/planners/rrt/src/RRT.cpp
RRTConnect: Motion planning start tree could not be initialized!
at line 267 in /tmp/binarydeb/ros-foxy-onpl-1.5.0/src/ompl/geonetric/planners/rrt/src/RRTConnect.
Parallelplan: :solve(): Unable to find solution by any of the threads in 0.012999 seconds
at line 138 in /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/tools/multiplan/src/ParallelPlan. cpp
RRTConnect: Motion planning start tree could not be {nitlalized!

CartestanTaskflow Fatlure
at line 51 in /hone/zhiyong00s/snpd_workspace/src/tesseract_planning/tesseract_process_nanagers/src/core/utils.cpp
RasterGlobalTaskflow Failure: Raster #4
at line 51 tn /hone/zhlyongo0ss/snpd_workspace/src/tesseract_planning/tesseract_process_nanagers/src/core/utils.cpp
GraphTaskflow Failure:
at Une 51 tn /hone/zhiyongo0ss/snpd_workspace/src/tesseract_planning/tesseract_process_nanagers/src/core/utils.cpp
RasterGlobalTaskflow Failure: to_end
at Une 51 in /hone/zhiyongo09s/snpd_workspace/src/tesseract_planning/tesseract_process_nanagers/src/core/utils.cpp
CartesianTaskflow Failure:
at Line 51 tn /hone/zhiyongo0ss/snpd_workspace/src/tesseract_planning/tesseract_process_nanagers/src/core/utils.cpp
RasterGlobalTaskflow Failure: Raster #3
at line 51 in /hone/zhiyong0ess/snpd_workspace/src/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
GraphTaskflow Fatlure:

Thanks in advance!!

marrts, 2023-03-12:
Make sure that the initial robot position is out of collision. The default position of the robot has the end effector in the collision, which you can change with the joint state publisher gui that launches when running. The application automatically uses the robots position as the start state

zhiyong-009, 2023-03-27:
thank you for the suggestion!

closed 2023-03-27 by zhiyong-009