ros-industrial-consortium/scan_n_plan_workshop #31: compatibility of generated mesh fields with PCL

SyZbidi, 2023-06-06:
Hello,

I'm currently using a similar setup to the scan_n_plan workshop under ROS1 where I run `industrial_reconstruction` with a realsense camera (`D435i`) then generate a `tool_path` using `noether` on the reconstructed mesh.

* realsense camera (version: [ros1-legacy](https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy))

* ros1 version of [industrial_reconstruction](https://github.com/sam-xl/industrial_reconstruction/tree/ros1) 

The problem is that the generated mesh has `double` x, y, z, nx, ny and nz while the PCL uses `float` and I get the error:

 
```
Failed to find match for field 'x'.

Failed to find match for field 'y'.

Failed to find match for field 'z'.

Error:   Vertex in PolygonMesh needs to have 3 elements only
``` 

I saw that you use ROS2 version of `realsense` which has an additional package `realsense2_camera_msgs` that doesn't exist on ros1, is that the missing key behind having an output mesh with `float` fields or are you using some kind of conversion in between reconstructing and `tpp`?

 

Thank you!

marip8, 2023-06-06:
> The problem is that the generated mesh has double x, y, z, nx, ny and nz while the PCL uses float and I get the error:
>
>```Failed to find match for field 'x'.
>
>Failed to find match for field 'y'.
>
>Failed to find match for field 'z'.
>
>Error:   Vertex in PolygonMesh needs to have 3 elements only```
>

Where do you get this message from? 

`industrial_reconstruction` saves the generated mesh to file using [an `open3d` function](https://github.com/sam-xl/industrial_reconstruction/blob/b338a3282fd4d81feed44c6d5f091d7696c52d73/industrial_reconstruction/nodes/industrial_reconstruction_node.py#L260). If you are trying to load this mesh file into PCL, then the error is presumably with the `open3d` mesh saving function. This [`open3d` issue](https://github.com/isl-org/Open3D/issues/1325) seems related. You could try updating the version of `open3d` to see if the issue is resolved. I personally have not run into any issues using this repository and `industrial_reconstruction` 2cf137d2 with `open3d` 0.15.2, but that has been in ROS2.

> I saw that you use ROS2 version of realsense which has an additional package realsense2_camera_msgs that doesn't exist on ros1, is that the missing key behind having an output mesh with float fields or are you using some kind of conversion in between reconstructing and tpp?

No, I don't think so. All of those messages seem to be for getting device specific meta-data.

SyZbidi, 2023-06-08:
> Where do you get this message from?

I got by launching `noether_examples plane_slicer_rastering_generator_demo.launch`

```
[pcl::PLYReader] /home/syrine/workspace/object_reconstruction_ws/src/reconstruction_ply/default.ply:15: property 'list uint8 uint32 vertex_indices' of element 'face' is not handled
Failed to find match for field 'x'.
Failed to find match for field 'y'.
Failed to find match for field 'z'.
Error:   Vertex in PolygonMesh needs to have 3 elements only
         at line 77 in /home/syrine/workspace/object_reconstruction_ws/src/noether/noether_conversions/src/noether_conversions.cpp
[ERROR] [1682331786.691695834]: Failed to read file /home/syrine/workspace/object_reconstruction_ws/src/reconstruction_ply/default.ply
[plane_slicer_generator_node-2] process has died [pid 429536, exit code 255, cmd /home/syrine/workspace/object_reconstruction_ws/devel/lib/noether_examples/plane_slicer_example_node __name:=plane_slicer_generator_node __log:=/home/syrine/.ros/log/007bcf26-e28a-11ed-b95f-a1c452af2903/plane_slicer_generator_node-2.log].
log file: /home/syrine/.ros/log/007bcf26-e28a-11ed-b95f-a1c452af2903/plane_slicer_generator_node-2*.log
```

I updated open3d to `0.15.2` and even `0.17.0` (I had 0.13.0) but I still have the fields as `double`.
Following the "work around" solution in the issue you linked got me to a same error different field -> vertex_indices:
```
property 'list uint8 uint32 vertex_indices' of element 'face' is not handled
```

Thank you for your reply, I'll look for a working work around, and eventually test it with ros2.

SyZbidi, 2024-02-15:
solved by using [plyfile](https://github.com/dranjan/python-plyfile/discussions/72#discussioncomment-6583212) for conversion. (ROS1 still)

closed 2024-02-15 by SyZbidi