ros-industrial-consortium/scan_n_plan_workshop #38: Installation Issues

mrunaljsarvaiya, 2023-07-07:
Are the installation steps in the main README up to date? Or are there additional dependencies that are not listed?

I've run into several errors while trying to setup this stack, the latest one being 

```
CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find MPI (missing: MPI_C_FOUND C)

      Reason given by package: MPI component 'C' was requested, but language C is not enabled.  

Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.22/Modules/FindMPI.cmake:1830 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/vtk-9.1/VTK-vtk-module-find-packages.cmake:397 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/vtk-9.1/vtk-config.cmake:150 (include)
  CMakeLists.txt:8 (find_package)
```

marip8, 2023-07-07:
> Are the installation steps in the main README up to date? Or are there additional dependencies that are not listed?

I believe all of the build dependencies are captured in `dependencies.repos` and `dependencies_tesseract.repos`. If you want to run one of the applications (`snp_blending`, `snp_automate_2022`), there are additional run-time dependencies that need to be cloned and built (e.g., Motoman support packages/driver, etc.)

What specific package failed and produced the error message above?

mrunaljsarvaiya, 2023-07-07:
> > Are the installation steps in the main README up to date? Or are there additional dependencies that are not listed?
> 
> I believe all of the build dependencies are captured in `dependencies.repos` and `dependencies_tesseract.repos`. If you want to run one of the applications (`snp_blending`, `snp_automate_2022`), there are additional run-time dependencies that need to be cloned and built (e.g., Motoman support packages/driver, etc.). Those should be documented in the respective package directories
> 
> What specific package failed and produced the error message above?

noether_tpp and noether_filtering both fail with the same error message above. For context, I'm building this in the ros2 humble docker container. Should that be supported?

marip8, 2023-07-07:
> For context, I'm building this in the ros2 humble docker container. Should that be supported?

We've only ever run SNP on Foxy. There has been a lot of work recently to update `tesseract_ros2` to work on 22.04 (Humble/Jammy), so we plan to support Humble here in the near future.

marrts, 2023-07-07:
Seems like [this](https://github.com/ros-industrial/noether/issues/189) issue on noether might be related

mrunaljsarvaiya, 2023-07-10:
> > For context, I'm building this in the ros2 humble docker container. Should that be supported?
> 
> We've only ever built/run the current version of SNP on Foxy. There has been a lot of work recently to update `tesseract_ros2` to work on 22.04 (Humble/Jammy), so we plan to support Humble here in the near future once that is complete.

Got it!

I started building the stack on a foxy docker image, after fixing a few dependencies I'm now running in noether_filtering failing because it can't find xmlrpcpp. I don't know if I'm missing something obvious but I haven't been able to find a reliable way to install xmlrpcpp. [This zip file](https://xmlrpcpp.sourceforge.net/faq.html) from this link has some information, but I'm running into issues building the make file. Any ideas on how to proceed?

marip8, 2023-07-10:
`sudo apt instal libxmlrpcpp`

marrts, 2023-07-10:
I was able to get noether building in humble with this branch: https://github.com/marrts/noether/tree/humble-devel.

I believe this should allow for the full scan_n_plan package to be run on humble now.

mrunaljsarvaiya, 2023-07-10:
Thank you! I've got the setup working now on foxy, I'll probably switch to humble and try out your branch later

after launching snp_automate_2022 through `ros2 launch snp_automate_2022 start.launch.xml`, I went through the steps in the rviz plugin:
1. get detailed scan 
2. generate tool path plan 
3. generate motion plan 

I get a failure on (3), logs attached below. I haven't changed any of the mesh/tool planning settings through the other gui that opens up. Is there any documentation on what the various  planning parameters are to help debug this?

```
[snp_motion_planning_node-7] Optimization results:
[snp_motion_planning_node-7] status: PENALTY_ITERATION_LIMIT
[snp_motion_planning_node-7] cost values: (0.0243942, 0.0182881, 0.0223221, 0.0134271, 0, 0, 0, 0, 0, 0, 0.0456933, 0.64492, 11.8282, 0.000378859, 0, 50.6274, 160.889, 958.131)
[snp_motion_planning_node-7] constraint violations: (1.7055e-07, 2.45747e-07, 2.94625e-07, 2.12127e-07, 4.25655e-10, 3.74105e-08, 1.83988e-09, 3.88354e-07, 2.1322e-06, 2.26175e-06, 2.95489e-06, 3.41135e-06, 0.000338777, 1.94892e-05, 9.78981e-06, 6.72639e-06)
[snp_motion_planning_node-7] n func evals: 74
[snp_motion_planning_node-7] n qp solves: 76
[snp_motion_planning_node-7] ==================
[snp_motion_planning_node-7] Info:    TRAJOPT motion planning failed (Failed to find valid solution) for process input: Raster #1
[snp_motion_planning_node-7] Debug:   worker 1 finished running TRAJOPT
[snp_motion_planning_node-7] Debug:   worker 1 ready to run Error Callback
[snp_motion_planning_node-7] Debug:   worker 1 finished running Error Callback
[snp_motion_planning_node-7] Debug:   worker 1 finished running Raster #1: Raster #1
[snp_motion_planning_node-7] Debug:   worker 1 ready to run From Start: from_start
[snp_motion_planning_node-7] Debug:   worker 1 ready to run CHECK_TASK_INPUT
[snp_motion_planning_node-7] Debug:   worker 6 ready to run Transition #1: Transition #1
[snp_motion_planning_node-7] Debug:   worker 7 ready to run CHECK_TASK_INPUT
[snp_motion_planning_node-7] Debug:   worker 1 finished running CHECK_TASK_INPUT
[snp_motion_planning_node-7] Debug:   worker 1 ready to run HAS_SEED_CHECK
[snp_motion_planning_node-7] Debug:   worker 7 finished running CHECK_TASK_INPUT
[snp_motion_planning_node-7] Debug:   worker 7 ready to run HAS_SEED_CHECK
[snp_motion_planning_node-7] Debug:   worker 7 finished running HAS_SEED_CHECK
[snp_motion_planning_node-7] Debug:   worker 7 ready to run SEED_MIN_LENGTH_CHECK
[snp_motion_planning_node-7] Debug:   worker 1 finished running HAS_SEED_CHECK
[snp_motion_planning_node-7] Debug:   worker 7 finished running SEED_MIN_LENGTH_CHECK
[snp_motion_planning_node-7] Debug:   worker 7 ready to run Error Callback
[snp_motion_planning_node-7] Debug:   worker 1 ready to run SEED_MIN_LENGTH_CHECK
[snp_motion_planning_node-7] Debug:   worker 7 finished running Error Callback
[snp_motion_planning_node-7] Debug:   worker 6 finished running Transition #1: Transition #1
[snp_motion_planning_node-7] Debug:   worker 1 finished running SEED_MIN_LENGTH_CHECK
[snp_motion_planning_node-7] Debug:   worker 1 ready to run Error Callback
[snp_motion_planning_node-7] Debug:   worker 1 finished running Error Callback
[snp_motion_planning_node-7] Debug:   worker 1 finished running From Start: from_start
[snp_motion_planning_node-7] Error:   GraphTaskflow Failure: 
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] Error:   RasterGlobalTaskflow Failure: Raster #1
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] Error:   GraphTaskflow Failure: 
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] Error:   RasterGlobalTaskflow Failure: Transition #1
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] Error:   GraphTaskflow Failure: 
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] Error:   RasterGlobalTaskflow Failure: from_start
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] [INFO] [1689018977.779919614] [snp_planning_server]: Failed to create motion plan
```

mrunaljsarvaiya, 2023-07-10:
Follow up -- I reduced some of the raster planner parameters, the status says CONVERGED but there's still a GraphTaskFlow failure 

```
[snp_motion_planning_node-7] Optimization results:
[snp_motion_planning_node-7] status: CONVERGED
[snp_motion_planning_node-7] cost values: (0.173069, 0.15701, 0.130848, 0.0981664, 0.0759429, 0.0481714, 0.0152581, 0, 0, 0, 0, 0.0436864, 0.373675, 9.64582, 0, 0, 0, 40.9813, 159.249, 936.242)
[snp_motion_planning_node-7] constraint violations: (2.03267e-06, 1.75855e-06, 1.56929e-06, 1.01447e-06, 4.51107e-07, 5.40044e-08, 1.10441e-07, 3.6829e-07, 2.34141e-08, 4.90062e-07, 4.51434e-06, 5.31683e-06, 8.98953e-06, 1.89281e-05, 2.10981e-05, 1.36696e-05, 1.25661e-05, 1.47266e-05)
[snp_motion_planning_node-7] n func evals: 53
[snp_motion_planning_node-7] n qp solves: 53
[snp_motion_planning_node-7] ==================
[snp_motion_planning_node-7] Debug:   Motion Planner process succeeded
[snp_motion_planning_node-7] Debug:   worker 5 finished running TRAJOPT
[snp_motion_planning_node-7] Debug:   worker 5 ready to run DISCRETE_CONTACT_CHECK
[snp_motion_planning_node-7] Debug:   worker 5 finished running DISCRETE_CONTACT_CHECK
[snp_motion_planning_node-7] Debug:   worker 5 ready to run Error Callback
[snp_motion_planning_node-7] Debug:   worker 5 finished running Error Callback
[snp_motion_planning_node-7] Debug:   worker 5 finished running Raster #1: Raster #1
[snp_motion_planning_node-7] Debug:   worker 5 ready to run From Start: from_start
[snp_motion_planning_node-7] Debug:   worker 5 ready to run CHECK_TASK_INPUT
[snp_motion_planning_node-7] Debug:   worker 3 ready to run Transition #1: Transition #1
[snp_motion_planning_node-7] Debug:   worker 1 ready to run CHECK_TASK_INPUT
[snp_motion_planning_node-7] Debug:   worker 5 finished running CHECK_TASK_INPUT
[snp_motion_planning_node-7] Debug:   worker 5 ready to run HAS_SEED_CHECK
[snp_motion_planning_node-7] Debug:   worker 1 finished running CHECK_TASK_INPUT
[snp_motion_planning_node-7] Debug:   worker 1 ready to run HAS_SEED_CHECK
[snp_motion_planning_node-7] Debug:   worker 5 finished running HAS_SEED_CHECK
[snp_motion_planning_node-7] Debug:   worker 1 finished running HAS_SEED_CHECK
[snp_motion_planning_node-7] Debug:   worker 1 ready to run SEED_MIN_LENGTH_CHECK
[snp_motion_planning_node-7] Debug:   worker 1 finished running SEED_MIN_LENGTH_CHECK
[snp_motion_planning_node-7] Debug:   worker 1 ready to run Error Callback
[snp_motion_planning_node-7] Debug:   worker 1 finished running Error Callback
[snp_motion_planning_node-7] Debug:   worker 5 ready to run SEED_MIN_LENGTH_CHECK
[snp_motion_planning_node-7] Debug:   worker 3 finished running Transition #1: Transition #1
[snp_motion_planning_node-7] Debug:   worker 5 finished running SEED_MIN_LENGTH_CHECK
[snp_motion_planning_node-7] Debug:   worker 5 ready to run Error Callback
[snp_motion_planning_node-7] Debug:   worker 5 finished running Error Callback
[snp_motion_planning_node-7] Debug:   worker 5 finished running From Start: from_start
[snp_motion_planning_node-7] Error:   GraphTaskflow Failure: 
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] Error:   RasterGlobalTaskflow Failure: Raster #1
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] Error:   GraphTaskflow Failure: 
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] Error:   RasterGlobalTaskflow Failure: Transition #1
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] Error:   GraphTaskflow Failure: 
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] Error:   RasterGlobalTaskflow Failure: from_start
[snp_motion_planning_node-7]          at line 51 in /snpd_ws/tesseract_planning/tesseract_process_managers/src/core/utils.cpp
[snp_motion_planning_node-7] [INFO] [1689019305.250033936] [snp_planning_server]: Failed to create motion plan
```

marrts, 2023-07-10:
Please open a new issue and let's move this discussion there as this no longer pertains the installation process.

closed 2023-07-10 by mrunaljsarvaiya