ros-industrial-consortium/scan_n_plan_workshop #44: SNP TPP Build Error

marip8, 2023-08-03:
I'm not sure if this is the right place for this. I tried running changes from this PR locally on my machine. I was able to compile fine, but when I run `ros2 launch snp_automate_2022 start.launch.xml`, the `snp_tpp_app` fails -

```
[snp_tpp_app-6] /home/srsidd//ws/install/snp_tpp/lib/snp_tpp/snp_tpp_app: symbol lookup error: /home/srsidd/ws/install/noether_gui/lib/libnoether_gui.so: undefined symbol: _ZNK4YAML4Node2asIiEET_v
[ERROR] [snp_tpp_app-6]: process has died [pid 43187, exit code 127, cmd '/home/srsidd/ws/install/snp_tpp/lib/snp_tpp/snp_tpp_app --ros-args --params-file /tmp/launch_params_kchfuebp'].
```

Not sure why. I tried doing some digging and it turns out if I comment out loading the mesh modifier ([L93 on noether_gui/src/tpp_pipeline_widget](https://github.com/ros-industrial/noether/blob/master/noether_gui/src/widgets/tpp_pipeline_widget.cpp#L93) - `mesh_modifier_loader_widget_->configure(config[MESH_MODIFIERS_KEY]);` and tool path modifier ([L110 on noether_gui/src/tpp_pipeline_widget](https://github.com/ros-industrial/noether/blob/master/noether_gui/src/widgets/tpp_pipeline_widget.cpp#L110) - `tool_path_modifier_loader_widget_->configure(config[TOOL_PATH_MODIFIERS_KEY]);` then the app runs fine without crashing.  I checked and the header files are all included. 

If it's better, I can open a separate issue, but considering it's part of this PR I thought I'd drop a comment here. Any thoughts? Have you run into this issue as well?

_Originally posted by @srsidd in https://github.com/ros-industrial-consortium/scan_n_plan_workshop/issues/22#issuecomment-1663148594_

marip8, 2023-08-03:
This issue should be addressed by #37. Are you using a version of scan-n-plan workshop that includes these changes? Also, you'll need to update your version of the `noether` repository to the hash listed in the `dependencies.repos` file. There is a corresponding update to `noether` that is captured in that hash

srsidd, 2023-08-03:
I'm using [this fork and branch ](https://github.com/marrts/noether/tree/humble-devel) that @marrts created for Humble. I did a meld to compare the two forks, and most of the changes in the fork I'm using are small fixes (Use `.hpp` instead of `.h`, use std instead of boost etc.). I don't see anything that can cause a linking error with yaml-cpp. Do you know what specific changes are in that hash that I can port over which I can fix? 

I have the other changes to include the correct `yaml-cpp` header files.

marip8, 2023-08-03:
AFAIK the port to the latest version of `tesseract` is totally independent of the `snp_tpp` package, so you should be able to rebase that branch onto the latest master branch (if it isn't already), pull the specified version of `noether`, rebuild and test.

srsidd, 2023-08-20:
Rebasing onto latest master and checking out the specific version didn't fix the error unfortunately.

However, after some more testing, I found something strange. When I compile my workspace in "Debug" mode (Specifying cmake arguments with colcon) the app does not crash, but when I compile in Release or `RelWithDebInfo` or `Release` the application crashes.