ros-industrial-consortium/scan_n_plan_workshop #46: Issue in installing rosdep keys

LuciferMorningStar0786, 2023-09-02:
Greetings all,
Me try to build this workshop in foxy 
When I use to try this command line to build the ROS2 dependenices

rosdep install --from-paths src --ignore-src -r -y

#issue    

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
tesseract_ros_examples: Cannot locate rosdep definition for [ros_testing]
tesseract_time_parameterization: Cannot locate rosdep definition for [ruckig]
tesseract_qt: Cannot locate rosdep definition for [gz-math7]
tesseract_collision: Cannot locate rosdep definition for [fcl]
snp_blending: Cannot locate rosdep definition for [joint_state_publisher_gui]
tesseract_task_composer: Cannot locate rosdep definition for [taskflow]
snp_automate_2022: Cannot locate rosdep definition for [joint_state_publisher_gui]
ros1_bridge: Cannot locate rosdep definition for [xmlrpcpp]
tesseract_motion_planners: Cannot locate rosdep definition for [ompl]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

Getting Isuses in Colcon build also

#issue          

Starting >>> noether_filtering
--- stderr: noether_filtering                                                           
CMake Error at CMakeLists.txt:17 (find_package):
  By not providing "Findxmlrpcpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "xmlrpcpp",
  but CMake did not find one.

  Could not find a package configuration file provided by "xmlrpcpp" with any
  of the following names:

    xmlrpcppConfig.cmake
    xmlrpcpp-config.cmake

  Add the installation prefix of "xmlrpcpp" to CMAKE_PREFIX_PATH or set
  "xmlrpcpp_DIR" to a directory containing one of the above files.  If
  "xmlrpcpp" provides a separate development package or SDK, be sure it has
  been installed.

---

Please help me out in installing this packages..
I want build Tessarct in ros2foxy for UR10e to do tool path generation on curved surface so please guide me through it... It will be more useful

marip8, 2023-09-05:
> Could not find a package configuration file provided by "xmlrpcpp" with any
of the following names:

You can fix this by running `sudo apt install libxmlrpcpp-dev`

Regarding the other issues:

> tesseract_ros_examples: Cannot locate rosdep definition for [ros_testing]`

Not critical

> tesseract_time_parameterization: Cannot locate rosdep definition for [ruckig]

Build from [source](https://github.com/pantor/ruckig)

> tesseract_qt: Cannot locate rosdep definition for [gz-math7]

Build with the cmake options `-DBUILD_STUDIO=OFF -DBUILD_RENDERING=OFF`

> tesseract_collision: Cannot locate rosdep definition for [fcl]

Build with the cmake option `-DTESSERACT_BUILD_FCL=OFF`

> snp_blending: Cannot locate rosdep definition for [joint_state_publisher_gui]
> snp_automate_2022: Cannot locate rosdep definition for [joint_state_publisher_gui]

`sudo apt install ros-<distro>-joint-state-publisher-gui`

> tesseract_task_composer: Cannot locate rosdep definition for [taskflow]
Install from the [ROS-I PPA](https://launchpad.net/~ros-industrial/+archive/ubuntu/ppa)

> ros1_bridge: Cannot locate rosdep definition for [xmlrpcpp]

`sudo apt install libxmlrpcpp-dev`

> tesseract_motion_planners: Cannot locate rosdep definition for [ompl]

`sudo apt install libompl-dev`

LuciferMorningStar0786, 2023-09-11:
Thankyou

closed 2023-09-11 by LuciferMorningStar0786