ros-industrial-consortium/scan_n_plan_workshop #50: Make `tcpSpeedLimiter` a tesseract plugin task

marrts, 2023-09-11:
The `tcpSpeedLimiter` function was originally implemented specifically for the HC10 which has restrictions on the tcp velocity which can cause motions to execute in an unexpected manner. This function should be a step in the planning pipeline that is only called for processes using the HC10, which is to say for `snp_automate_2022`. The currently implemented function can be found [here](https://github.com/ros-industrial-consortium/scan_n_plan_workshop/blob/27be614fd107788e706fd799e192d273dc66a4e2/snp_motion_planning/src/planning_server.cpp#L254) and examples of implementing a custom task for a tesseract planning pipeline can be found [here](https://github.com/ros-industrial-consortium/scan_n_plan_workshop/tree/27be614fd107788e706fd799e192d273dc66a4e2/snp_motion_planning/src/plugins/tasks).

closed 2024-01-05 by marrts

rjoomen, 2024-01-08:
Is it an idea to add this to main tesseract (or tesseract_planning)?

marrts, 2024-01-08:
I wasn't planning on it. This feels like a robot specific thing (it stemmed from issues we were having with motion execution on the Motoman HC10). I also like having example(s) like this so people know how to make their own custom plugins.