ros-industrial-consortium/scan_n_plan_workshop #54: Create docker file/image for ROS bridge

marip8, 2023-10-09:
Per the suggestion of @gavanderhoorn, we should provide a Docker image (and probably Docker file) for the ROS bridge that supports the required topics, services, and actions required by the SNP application. The entrypoint for this script should probably be the bridge launch command (`ros2 run ros1_bridge dynamic_bridge --bridge-all-1to2-topics`).

Note the bridge install instruction from [here](https://github.com/ros-industrial-consortium/scan_n_plan_workshop/tree/3e2ee90b4e9b638af9a1e807ccf4df85588361e6/snp_automate_2022) denotes a branch of the bridge (`action_bridge`) that has since been removed or overwritten

marrts, 2023-10-09:
The commit history is lost somehow, but [here](https://github.com/DavidSpielman/ros1_bridge) is the exact state of that branch pulled from the demo computer we were running for the SNP demo.

marip8, 2023-10-18:
@collinthornton is looking into this. I think we'll need to install the following packages in both ROS distros:

- `actionlib_msgs`
- `std_msgs`
- `sensor_msgs`
- `control_msgs`
- `industrial_msgs`
- `rosgraph_msgs`
- `trajectory_msgs`
- `tf2_msgs`
- `geometry_msgs`

marip8, 2023-11-07:
Docker files for the `action_bridge` version of the `ros1_bridge` were added [here](https://github.com/ros-industrial/ros1_bridge/tree/action-bridge/docker). A pre-built Docker image can be downloaded via `docker pull ghcr.io/ros-industrial/ros1_bridge:noetic-foxy`