ros-industrial-consortium/scan_n_plan_workshop #116: Getting Error while doing vcs import < src/scan_n_plan_workshop/dependencies_tesseract.repos

arulfrances, 2024-05-20:
Team,

I have cloned the repo  and while doing vcs import, I am getting the following error:

[image]

Please guide me in the right direction.

P.S: My internet connection is perfectly stable.

What I tried:
git fetch --unshallow
git fetch --depth=2147483647

But it didn't help! 

Thanks

marip8, 2024-06-03:
The process should be:

```bash
cd <snp_workspace>
vcs import src < src/scan_n_plan_workshop/dependencies_tesseract.repos
```

Some of the repos like `tesseract` are fairly large, so you could also try using a shallow clone to make the process go faster (i.e., `vcs import --shallow src < src/scan_n_plan_workshop/dependencies_tesseract.repos`)

Closing; this was added to the README in #122

closed 2024-06-03 by marip8

arulfrances, 2024-06-03:
Thanks a lot @marip8  for your support as always!

After adding the shallow command, I was able to build the App without any issues!

Keep up the great work on supporting the community!