ros-industrial-consortium/scan_n_plan_workshop #118: Mesh Loading Failure

marrts, 2024-05-24:
If you pass a mesh to the motion planner and there is no mesh at that filepath it will print an error, but just continue with the motion plan. This could lead to unexpected behavior and motion plans that are not collision free.
```
[snp_motion_planning_node-8] Error:   Could not load mesh from "/tmp/results_mesh_asdfsa2.ply": Unable to open file "/tmp/results_mesh_asdfsa2.ply".
```

marrts, 2024-05-24:
I think it would probably make the most sense to separate out the adding the mesh process from the motion plan call. This would give us more flexibility in motion planning to either require a mesh or not. It would also potentially enable adding multiple meshes if the scenario ever came up.

Moonwanlite, 2024-06-25:
> I think it would probably make the most sense to separate out the adding the mesh process from the motion plan call. This would give us more flexibility in motion planning to either require a mesh or not. It would also potentially enable adding multiple meshes if the scenario ever came up.

How would you separate out adding the mesh process?

marrts, 2024-06-25:
> How would you separate out adding the mesh process?

1. Add a new service definition in the `snp_msgs` package.
2. Add a new service to the planning server that has the sole purpose of adding a mesh to the environment.
3. Add a new behavior tree node to call this service
4. Modify the behavior tree pipelines to call this service where appropriate
5. Remove references to adding a mesh in:
    - The motion planning service callback
    - Motion planning service msg definition
    - behavior tree node related to motion planning