ros-industrial-consortium/scan_n_plan_workshop #126: Upon following the initial steps for building the scan-n-plan package in a fresh workspace, a build error is thrown.

wglockner, 2024-06-19:
`Building ISP
Building TOTG
Building Ruckig
CMake Error at ruckig/CMakeLists.txt:1 (find_package):
  By not providing "Findruckig.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ruckig", but
  CMake did not find one.

  Could not find a package configuration file provided by "ruckig" with any
  of the following names:

    ruckigConfig.cmake
    ruckig-config.cmake

  Add the installation prefix of "ruckig" to CMAKE_PREFIX_PATH or set
  "ruckig_DIR" to a directory containing one of the above files.  If "ruckig"
  provides a separate development package or SDK, be sure it has been
  installed.
`
I have sourced /opt/ros/foxy/setup.bash prior to building, and I have tried installing each package individually as well.  The tesseract_planning package fails to build in both cases.  I have solved the problem locally by sourcing ros_industrial/devel/setup.bash after sourcing /opt/ros/foxy/setup.bash.  This was not part of the install instructions.

marrts, 2024-06-19:
`ruckig` is listed as a dependency in the `tesseract_time_parameterization` [package.xml.](https://github.com/tesseract-robotics/tesseract_planning/blob/4db71a632d3c9264d7a60f6ce1c2317d7cb12295/tesseract_time_parameterization/package.xml#L16) It appears that `tesseract_time_parameterization` is the package that failed to build for you. This dependency (`ros-foxy-ruckig`) should be automatically installed when you run the following command from the top level of your workspace:
```
rosdep install --from-paths src --ignore-src -r -y
```
I'm not sure what sourcing your local workspace would accomplish to resolve this, unless you were also building ruckig from source which you would've had to go out of your way to do. Also, it appears that you might be sourcing a ROS 1 workspace? Not sure why you are sourcing something from a devel directory. It's possible you have the ruckig library installed for ros noetic and you added all your ROS 1 libraries to your library path and then `tesseract_time_parameterization` built by linking against the ROS 1 version of ruckig.

To ensure you have `ruckig` properly installed you can run:
```
apt list --installed | grep ruckig
```
You should see something like:
```
ros-foxy-ruckig/focal,now 0.9.2-1focal.20230527.033136 amd64 [installed]
```
If this is installed then that indicates ROS 2 might not have been properly sourced. If it's not installed then maybe the `rosdep` command was run improperly. If you only see `ros-noetic-ruckig` then that lines up with my above theory.

wglockner, 2024-07-12:
Thank you for your response, I was able to continue with the install but only after installing a significant amount of additional dependencies.  Many of the gz-*** packages are not installed after running the commands in the README and so manual installation seems to be the only option that worked for me.

marip8, 2024-07-12:
It's not entirely obvious, but the `gz` packages are required by the `tesseract_qt` rendering and studio modules which aren't needed by SNP. If you compile your workspace with the CMake options `-DTESSERACT_BUILD_RENDERING=OFF -DTESSERACT_BUILD_STUDIO=OFF` then it won't build those modules or require the `gz` dependencies. I thought we put this in our SNP documentation somewhere, but maybe not

wglockner, 2024-07-12:
Thank you, I was able to build with these changes.

closed 2024-07-12 by wglockner