ros-industrial-consortium/scan_n_plan_workshop #152: KinematicGroup::calcInvKin Assert Failure

Simon248, 2025-01-15:
Hello

I'm facing this issue when launching a plan request:
```
tesseract/tesseract_kinematics/core/src/kinematic_group.cpp:130: 
tesseract_kinematics::IKSolutions tesseract_kinematics::KinematicGroup::calcInvKin(const KinGroupIKInputs&, const Eigen::Ref<const Eigen::Matrix<double, -1, 1> >&) const: 
Assertion `std::abs(1.0 - tip_link_pose.pose.matrix().determinant()) < 1e-6' failed
```

Context:

I first faced this issue trying to build my own docker image of snp_workshop with binded source (to deal with that: [link](https://github.com/tesseract-robotics/tesseract_planning/discussions/251) ). 
So i followed the build setup in snp_workshop readme a faced this issue.

The URDF, SRDF (including IK plugin yaml setup) and Toolpath i use are working well on another setup where i use the image ghcr.io/ros-industrial-consortium/scan_n_plan_workshop:humble-5.0

I can confirm that trying to follow [snp_automate_2023 demo](https://github.com/ros-industrial-consortium/snp_automate_2023) local build, i face the same issue when planning.

I can confirm that using docker setup from the same repo [snp_automate_2023 demo](https://github.com/ros-industrial-consortium/snp_automate_2023) i dont have the problem.

Not sure if it comes from my system or from the tesseract packages. But as i tried to build it from scratch on a fresh image it could be the second, so i'm posting this here.

BR and thanks.

marip8, 2025-01-15:
This comes from the inverse kinematics function in `tesseract_kinematics`, and it's essentially making a check that the orientation of the target Cartesian pose is valid. Usually you get this assertion when the columns of the rotation matrix are not exactly orthogonal. How are you generating target poses for `tesseract`/SNP?

Simon248, 2025-01-29:
Thanks for your reply, 

I dont think the problem comes from my target/toolpaths. they are generated using opencascade [gp_Ax3](https://dev.opencascade.org/doc/refman/html/classgp___ax3.html#details) which ensure orthogonality.

And moreover, same toolpath is planned fine using snp docker image. The problem only occur when using snp installation on my host.

I made different image with same version of tesseract and I dont have that problem, so it's probably linked to my host and some dependencies...

I can't explain it now and i dont work on my host anymore (only containers).

closed 2025-01-29 by Simon248